rm-hull/luma.examples #157: Tweepy code

tiellas, 2022-10-26:
Getting a few issues running the tweepy example. Tweepy has been updated, and StreamListener was merged to Stream. But even after updating that in my code I'm still getting so errors. Anyone think they can take a look at the tweet_scroll.py code and see how to get it to work?

thijstriemstra, 2022-10-26:
>  after updating that in my code I

Can you share the code?

>  I'm still getting so errors.

What errors? what version of python/OS?

tiellas, 2022-10-26:
`from __future__ import unicode_literals

consumer_key = "7PVZ1E1PhwHDopDc0qoGuojPP"
consumer_secret = "zWISdkC6laOrMwzGQCUaIhxjJpmhnCNbXXQPR7GHKe0nja5XQp"
access_token = "1296930319648653312-kC9DacUeaJJwiOpYwXd75AOKqeNBMp"
access_token_secret = "JMaRIn8u2Wk299judq2X8NId1EcUes7X8TxlmZggHWIEi"

search_terms = ['bladee']

import sys
import time
from pathlib import Path
from PIL import ImageFont

try:
    from Queue import Queue
except ImportError:
    from queue import Queue

from demo_opts import get_device
from luma.core.render import canvas
from luma.core.virtual import viewport

try:
    import tweepy
except ImportError:
    print("The tweepy library was not found. Run 'sudo -H pip install tweepy' to install it.")
    sys.exit()

def make_font(name, size):
    font_path = str(Path(__file__).resolve().parent.joinpath('fonts', name))
    return ImageFont.truetype(font_path, size)

def scroll_message(status, font=None, speed=1):
    author = u"@{0}".format(status.author.screen_name)
    full_text = u"{0}  {1}".format(author, status.text).replace("\n", " ")
    x = device.width

    # First measure the text size
    with canvas(device) as draw:
        w, h = draw.textsize(full_text, font)

    virtual = viewport(device, width=max(device.width, w + x + x), height=max(h, device.height))
    with canvas(virtual) as draw:
        draw.text((x, 0), full_text, font=font, fill="white")
        draw.text((x, 0), author, font=font, fill="yellow")

    i = 0
    while i < x + w:
        virtual.set_position((i, 0))
        i += speed
        time.sleep(0.025)

class listener(tweepy.Stream):

    def __init__(self, queue):
        super(listener, self).__init__()
        self.queue = queue

    def on_status(self, status):
        self.queue.put(status)

device = get_device()
auth = tweepy.OAuthHandler(consumer_key, consumer_secret)
auth.set_access_token(access_token, access_token_secret)
api = tweepy.API(auth)
queue = Queue()

if device.height >= 16:
    font = make_font("code2000.ttf", 12)
else:
    font = make_font("pixelmix.ttf", 8)

try:
    stream = tweepy.Stream(auth=api.auth, listener=listener(queue))
    stream.filter(track=search_terms, is_async=True)  # noqa: W606

    try:
        while True:
            status = queue.get()
            scroll_message(status, font=font)
    except KeyboardInterrupt:
        pass

finally:
    stream.disconnect()`

**### This is the full code^^^**

`Traceback (most recent call last):
  File "/home/tiellas/luma.examples/examples/tweet_scroll.py", line 108, in <module>
    stream = tweepy.Stream(auth=api.auth, listener=listener(queue))
  File "/home/tiellas/luma.examples/examples/tweet_scroll.py", line 89, in __init__
    super(listener, self).__init__()
TypeError: __init__() missing 4 required positional arguments: 'consumer_key', 'consumer_secret', 'access_token', and 'access_token_secret'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tiellas/luma.examples/examples/tweet_scroll.py", line 119, in <module>
    stream.disconnect()
NameError: name 'stream' is not defined`

### **This is the error^^**

thijstriemstra, 2022-10-26:
> Sorry it formatted weird, im not proficient on here. 

I'll fix that..

tiellas, 2022-10-26:
> > Sorry it formatted weird, im not proficient on here. 
> 
> 
> 
> I'll fix that..

Great thanks, I noticed on mobile GitHub app the comment formatted fine. And I am using python3 on default raspberry pi OS (aka raspian bullseye I belive)