rm-hull/luma.examples #158: Deprecated Pillow Commands

toldcodger, 2022-11-29:
Hi,
I am not a Python expert but I have tried to solve this myself (unsuccessfully) before contacting you.
Several Luma LCD examples throw up errors about upcoming deprecated Pillow commands.  The unaltered examples run successfully until I try to change the textsixe line as described below.
Below is an example of one such error in font_awsome.py  

I edited the script to work with an ST7735 and renamed the file to font_awesomeSPI.py to reflect this.

  w, h = draw.textsize(text=code, font=font)
/home/*****/luma.examples/examples/font_awesomeSPI.py:162: DeprecationWarning: textsize is deprecated and will be removed in Pillow 10 (2023-07-01). Use textbbox or textlength instead.
  w, h = draw.textsize(text=code, font=font)

If I swap textsize for textbbox I get the error below:

    w, h = draw.textbbox(text=code, font=font)
TypeError: textbbox() missing 1 required positional argument: 'xy'

If I swap textsize for textlength I get the error below:         

    w, h = draw.textlength(text=code, font=font)
TypeError: cannot unpack non-iterable float object

any help would be gratefully appreciated

Regards

Dave

thijstriemstra, 2022-11-29:
> Several Luma LCD examples throw up errors about upcoming deprecated Pillow commands.

This is probably related to fact you're using an older version of luma.core and luma.lcd. Try upgrading them, e.g.

```
pip3 install -U luma.lcd
```

thijstriemstra, 2022-11-29:
Ah nevermind, you're right, the examples still need to updated for the latest Pillow, e.g. https://github.com/rm-hull/luma.examples/search?q=textsize

toldcodger, 2022-11-29:
Hi Thijs,

Thanks for the info.  If we are just waiting for an update, and you don’t need to keep it open for your own purposes, please feel free to close this.

Thanks again, it’s good to know I’m not going crazy 😂😂😂

Regards

Dave

PR #162 (2023-08-03): Replace usage of deprecated Pillow methods
found following:

```console
$ grep -Rn "textsize" examples/
./tweet_scroll.py:70:        w, h = draw.textsize(full_text, font)
./image_composition.py:29:            w, h = draw.textsize(text, font)
./demo.py:43:    size = draw.textsize('World!')
./font_awesome.py:152:                w, h = draw.textsize(text=code, font=font)
./greyscale.py:40:                size = draw.textsize("greyscale")
./welcome.py:167:            size = draw.multiline_textsize(t, font)
./welcome.py:170:                size = draw.multiline_textsize(t, font)
./hotspot/common.py [... cut]