rm-hull/luma.examples #159: Example IP, Temp, Memory, Disk

redsix666, 2022-12-29:
Hi

I have a new Mini Tower for my Pi and it shows an example in a picture with local IP, Temp, Memory and Disc information.
I tried a lot examples but couldn't find the right one. Is it an additional example that i need?

Thanks for answering

yoyojacky, 2023-05-12:
it seems like you don't install the libraries properly, just re-execute the install.sh script to install the libraries. 
or execute ：
```bash
cd luma.examples/
sudo -H pip install -e .
```

matiasmtl, 2023-07-09:
I have the same problem. minitower_oled.service points to sysinfo.py in example but there's not such file there. The closest is sys_info.py When correct it to that one I get temp, ram but not the IP as shown in the example.

rlaace423, 2023-08-27:
@redsix666 @matiasmtl 
I think I bought exact same NAS Kit like yours.
I wrote similar examples for my own use, and made PR of it.

Check this [PR](https://github.com/rm-hull/luma.examples/pull/163)

closed 2023-09-25 by thijstriemstra