rm-hull/luma.examples #160: Displaying both static and scrolling text

dineshimatrix, 2023-03-16:
Hi,

First of All, a very great applause and sincere thanks to you for the library.

I've an MAX7219 display which was arranged in a 32x64 fashion(width=64 pixels and height =32 pixels). I'm able to display static text in every row and now i need to display static text in first 3 rows and a scrolling message in last row. But i dont know how to handle this. Trying the loops is not a good idea i think, since i assume this kind of scenarios might be handled already in the library. 

Please give me any hint to solve this issue.

javafiles, 2023-04-21:
Hi,

Did you find a solution for this? I need to display both static and scrolling text;

Thank you

dineshimatrix, 2023-05-15:
> Hi,
> 
> Did you find a solution for this? I need to display both static and scrolling text;
> 
> Thank you

Hi @javafiles 

I got the idea. While scrolling the text i updated the static code also.

ex:

loop {
static text
move scroll content by one pixel
}

loop until all the scrolling text has been passed out of canvas