rm-hull/luma.examples #164: Static noise on oled I2C SH1106 display

bartvdg, 2023-10-13:
When running pi_logo.py from the examples I get the device info but the screen show for 5/6 static noise and only showing from the top a part (1/6) of the raspberry pi logo scrolling right to left?
I did follow all instructions to install libraries etc... this went fine w/o any errors.

**device info:**
_Version: luma.oled 3.13.0 (luma.core 2.4.1)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64_

KicksTV, 2024-03-30:
I have the same issue. Did you ever solve the issue?

thijstriemstra, 2024-03-30:
@KicksTV @bartvdg can you post any pictures of this issue? what OS and python version?

KicksTV, 2024-03-30:
Thanks for assisting @thijstriemstra.
Python 3.11.2
Debian GNU/Linux 12 (bookworm)
Raspberry Pi 4 Model B Rev 1.4
[image]

thijstriemstra, 2024-03-30:
you don't happen to have an identical screen laying around to reproduce the issue on that screen? I've used these screens without problems and my guess this one might be broken.

bartvdg, 2024-04-02:
Hi,
I have **sh1106** model oled display and using this command to display from py program:
python3 program.py --display sh1106

It took a while to figure things out since the documentation from  az-delivery (vendor) was crap.

You need to install/run following on your pi (I'm running on ZeroW)
**A. Run first three commands if Python3, pip3 and git are not installed**
sudo apt-get install python3-dev libffi-dev libssl-dev python3-pil libjpeg-dev zlib1g-dev libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff5 -y
sudo apt-get install python3-rpi.gpio python3-pip -y
sudo apt-get install git -y
**B. Install libs**
sudo -H pip3 install luma.oled
**C. Create folder "oled"**
sudo mkdir oled
cd oled
**C. get the examples**
sudo git clone https://github.com/rm-hull/luma.examples
**D. Test**
python3 demo.py --display [controller]
example:  python3 demo.py --display sh1106
supported controllers: 
 'ssd1306', 'ssd1309', 'ssd1322', 'ssd1362', 'ssd1322_nhd', 'ssd1325', 'ssd1327', 'ssd1331', 'ssd1351', 'sh1106', 'sh1107', 'ws0010', 'winstar_weh', 'linux_framebuffer'