rm-hull/luma.examples #166: Module(s) not found because of wrong installation

jvherck, 2023-12-03:
The installation guide on the homepage https://github.com/rm-hull/luma.examples#readme says to use 
```sh
$ sudo -H pip install -e .
```
to install the dependencies. I tried and I got the error `ModuleNotFoundError: no module named luma`.
I was able to resolve this by using 
```sh
$ sudo -H pip3 install -e .
```
instead. When trying to run something now, it does work. So please update your documentation please :)

dg1kwa, 2023-12-15:
I have this (Raspbian 12 64bit):

pi@raspberrypi:~/luma.examples $ sudo -H pip3 install -e .
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
pi@raspberrypi:~/luma.examples $

thijstriemstra, 2023-12-15:
> So please update your documentation please :)

See #167 

@dg1kwa please try the updated instructions, temporarily located at https://github.com/rm-hull/luma.examples/tree/env-doc#installation-instructions until that pull request is merged.

dg1kwa, 2023-12-15:
> > So please update your documentation please :)
> 
> See #167
> 
> @dg1kwa please try the updated instructions, temporarily located at https://github.com/rm-hull/luma.examples/tree/env-doc#installation-instructions until that pull request is merged.

this work finde with raspbian 12 64bit. Thank you very much for very fast fix the problem.

PR #167 (2025-07-02): doc: use virtual env for installation
fixes #166
commit 63da44e2 (2025-07-02): doc: use virtual env for installation (#167)

* doc: use virtual env for installation

fixes #166

* fix markup

---------
closed 2025-07-02 by rm-hull