rm-hull/luma.examples #180: When using a st7789 display (135x240), V-Offset anf H-Offset are not working

thejunglejim, 2025-02-13:
Hi,

I'm using a st7789 display which is 135x240. It is this one https://www.adafruit.com/product/4393 

Using your examples with the following parameters to test it:
`python3 bounce.py --display=st7789 --interface=spi --spi-bus-speed=52000000 --gpio-data-command=25 --gpio-backlight=22 --backlight-active=high --width=240 --height=135`
I can get the screen to display but part of the image is moved off to the top and left. See attached image.

[image]

I tired using `--h-offset` and `--v-offset` with different values but it does not seem to make a difference. 

On other drivers, you normally specific the following offsets:  `offset_left=40 offset_top=53`

I not sure how to get the display to align correctly??

thijstriemstra, 2025-02-14:
what kind of device and OS are you using?

thejunglejim, 2025-02-14:
I'm using a Raspberry Pi 2 Zero  W  with the latest Pi OS