rm-hull/luma.examples #181: Meshtastic android app. not able to text.

Allanusa, 2025-02-15:
I am new LoRa and the Meshtastic app. I started at the Meshtastic.org website to find the step to begin. However, I first learned about the LoRa off of Youtube. I watched several videos to find out if this platform will work for my project "transmitting tempters and amps data". In setting up the LoRa to learn about how it woks. I installed the Meshtastic app. Found that I can change the setting with the phone app. I was not able the text from the app to the LoRa device. I have tried the Heltec wireless stick, Heltec esp32 v3 wifi/bluetooth, and the Heltec mes nod t114 nFf52840+SX1262. I have tried two of each devices. Updated supporting software unload and reloaded. Tried three different USB cables on every Com port. Three different android devices, upload, unload and reload on all. Two different computers. Used Arduino and VS Code to run some test examples will little luck. Never find any place that I am able to use the serial monitor with Arduino, VS Code and Meshtastic Web serial monitor to communicate with any LoRa module. With this much failure I am sure I am doing something wrong. Any pointes?

thijstriemstra, 2025-02-15:
wrong project? this is not related to luma.examples

Allanusa, 2025-02-15:
I apologize for that. I am learning how to navigate Github. I was going to move it to a different repository. Not found one witch my issue would fit.

closed 2025-02-15 by thijstriemstra